valzzu/meshtastic-pcbs #1: Firmware: which one to choose?

hayschan, 2024-12-30:
Love your design! Great work.

Which firmware should I choose from the official flasher? Or do I need to change the pins and compile myseslf through PlatformIO?

valzzu, 2024-12-30:
It's the promicro diy Variant in the flasher

closed 2024-12-30 by valzzu

hayschan, 2024-12-30:
Thanks a lot. Great work. 

I've recommended to a lot of people in the MeshCN.net community. 

We love your work!! 🚀

valzzu, 2024-12-30:
awesome :) hope to see them in the wild xd

valzzu, 2024-12-31:
@hayschan btw what frequency do u ppl use over there?

hayschan, 2025-01-02:
> @hayschan btw what frequency do u ppl use over there?
> 
> Since the module currently on it is for 915 and 868

We use 470 MHz. We will have to change the BoM to use a 470 MHz module instead.

valzzu, 2025-01-02:
> > @hayschan btw what frequency do u ppl use over there?
> > 
> > Since the module currently on it is for 915 and 868
> 
> We use 470 MHz. We will have to change the BoM to use a 470 MHz module instead.

https://www.cdebyte.com/products/E22-400M30S/1

:)