valzzu/meshtastic-pcbs #6: Project files

shurillu, 2025-02-05:
Hello, 

   nice build, I want to customize some things, like pullup resistor for user/reset button , GPS port, maybe rotary encoder + buzzer.
Whould you share the file projects?

Thanks, regards. 

Stefano

valzzu, 2025-02-05:
If u want, u can re create it from the schematic. Id rather not share the project files .
Ill add a resistor to the reset button next weekend when im home.

But for gps, rotary encoder and buzzer u could just directly solder to the promicro.

NomDeTom, 2025-02-06:
I don't think the reset needs a pullup? I've certainly never added one...

User button is set to pullup internally, too.

valzzu, 2025-02-06:
Tru 😅 and theres only reset.

shurillu, 2025-02-06:
> I don't think the reset needs a pullup? I've certainly never added one...
> 
> User button is set to pullup internally, too.

There are several opinions about the pullup resistor on reset pin (and generally on input pins): I saw schematics with and without. Even in the Nordic forum, Nordic techies have different opinions.
As a rule of thumb, if you put it by hardware, the pin have certainly the pullup resistor.
But I'm new in PCB design, so forgive me ;-)

I'm going to close the thread, thank you for your response/feedback!

Regards, 

Stefano

closed 2025-02-06 by shurillu