valzzu/meshtastic-pcbs #8: Document: Battery does not charge from USB

wbrucem, 2025-02-05:
Mention that the battery does not receive charging from the USB port, solar is required to charge the battery.

wbrucem, 2025-02-05:
I wonder whether a simple diode could send USB power into the charging circuit...?

valzzu, 2025-02-06:
Wait rly? Did u try turning the switch on?

wbrucem, 2025-02-06:
You are correct, it is charging very slowly (appears to be just over 100mA). Sorry for my mistake.

wbrucem, 2025-02-06:
FYI, I calibrated the battery voltage reading:
ADC_Multiplier Override = 1.676

I believe the default value is 1.85

valzzu, 2025-02-07:
> FYI, I calibrated the battery voltage reading: ADC_Multiplier Override = 1.676
> 
> I believe the default value is 1.85

Hmmm..... U shouldn't need to change it 🤔 well if its more accurate then no clue.

closed 2025-02-10 by valzzu