valzzu/meshtastic-pcbs #9: Need battery cutoff @3.0v

wbrucem, 2025-02-05:
Most battery protection circuits allow the battery to discharge to about 2.5v, but the nRF52 halts at around 2.8v. Even after recharging, the MCU is still halted, requiring manual intervention to restore operation. Consider adding a circuit to cut off power at 3.0v to ensure the device will automatically recover after full battery discharge.

valzzu, 2025-02-06:
Theres a ic that cuts the power at 2.9v

wbrucem, 2025-02-10:
I did some experiments with my bench supply connected to the battery port.

Power consumption at idle is about 0.07 watt. This jumps to around 3.6 watts during transmission. Current limiting would trigger during transmission when set to 1.2 amps, but not at 1.25.

I was unable to verify power being cut off at 2.9v. In fact, the device continued to operate down to 2.08 volts! The 5v boost starts to drop around 2.45v.

Also, THANK YOU SO MUCH for sharing your design. It is precisely what I was looking for.

valzzu, 2025-02-11:
> I did some experiments with my bench supply connected to the battery port.
>
> Power consumption at idle is about 0.07 watt. This jumps to around 3.6 watts during transmission. Current limiting would [...]

What version of the board do u have? There was an issue with the protection module but it was fixed in v0.3.1.

wbrucem, 2025-02-12:
My boards are marked "V0.3.1", downloaded January 19.

Testing with a 2000mAh battery, it ran for 2 days plus about 7 hours while connected to my local rural mesh (active but not dense traffic). No doubt would discharge a bit more quickly in a busy urban environment.

valzzu, 2025-02-12:
> My boards are marked "V0.3.1", downloaded January 19.
> 
> Testing with a 2000mAh battery, it ran for 2 days plus about 7 hours while connected to my local rural mesh (active but not dense traffic). No doubt would discharge a bit more quickly in a busy urban environment.

That should have the issue fixed 🤔

NomDeTom, 2025-02-12:
Hiya @wbrucem , I've been doing some digging with @valzzu and it looks like the battery protection IC wasn't fixed until 0.3.2 of the board:

[image]

Unfortunately it's not an easy fix if the IC is already soldered down, but if you have a spare board, or you're comfortable desoldering it and then putting it back then you could sever the traces around the bottom pad:

[image]

Let us know how you get on!

valzzu, 2025-02-12:
> Hiya @wbrucem , I've been doing some digging with @valzzu and it looks like the battery protection IC wasn't fixed until 0.3.2 of the board:
> 
> [image]
> 
> Unfortunately it's not an easy fix if the IC is already soldered down, but if you have a spare board, or you're comfortable desoldering it and then putting it back then you could sever the traces around the bottom pad (and that via!):
> 
> [image]
> 
> Let us know how you get on!

Welll... It has via in the middle soooo..... You would have to make sure its not soldering onto that also.

valzzu, 2025-02-12:
Nvm, it should be fine 😅 i got scared and looked at the wrong ic i think

wbrucem, 2025-02-12:
I am not equipped to work with surface mount chips, but I think I can live with it as is. The 2 units I have tested so far are running until the battery's built-in protection circuit shuts off power at 2.5v. Glad to see the mystery is solved anyway.  Thank you.

valzzu, 2025-02-12:
Imma close this for good now, hope u like the node 🙃
Please share it with friends or something 🤣 anyone good luck.

closed 2025-02-12 by valzzu