valzzu/meshtastic-pcbs #10: Reversed Polarity

dswinton, 2025-03-08:
Hey There @valzzu ! 

First off - thanks for creating this and making it open source!  I have a few of the 0.3.3 version of Washtastic and they are awesome!  I'm rolling them out in our local mesh as our main repeaters, to replace the RAK boards we have there now.  You've created what I think is one of the best boards for Meshtastic out there!

I know this 110% my own fault - but I've managed to kill 2 of the boards by connecting the battery with reverse polarity.  How I did it was using the through-hole connectors to put the JST connector on (rather than surface mount connectors as you have in the photos).  I didn't realise that the through hole connectors are the opposite polarity to the surface mount connectors, so I just followed the +/- markings on the board.

So - one bit of feedback to noob-proof boards against people like me... it might be a good idea to either potentially switch the polarity of the through-hole connectors, or add some markings on them to show that they're reversed.  

On both boards, something seemed to get very hot just near the battery connector before releasing magic smoke, but I couldn't quite tell what component it was.  As far as If you wouldn't mind, could you please point me in the right direction as to which components I might have blown and need to replace..?  I've determined the 1k resistor right near the connector is still ~1K ohms, and I'm reading the battery voltage on the Batt +/- pins on the NRF board... beyond that I'm not sure what to test next!

Cheers!

valzzu, 2025-03-08:
i'm happy to hear my stuff are good :) i never thought I would create something that ppl would actually deploy out there in the world .

The battery plug has always been correct on the surface and through hole connections.
solar plug had a mistake that I corrected in 0.3.3b, the through hole was reversed.

could u send a pick of the bottom part of the PCB?

[image]

valzzu, 2025-03-09:
That is quite interesting, there shouldn't be any way for jlc to reverse the connections.

valzzu, 2025-03-09:
Plus nothing on those photos look broken.
Does it work if u try powering it on?

valzzu, 2025-03-09:
@dswinton could i get photo of just the connector?

valzzu, 2025-03-09:
like the a photo where i can see the connector or where it goes and the surrounding area

valzzu, 2025-03-09:
okay, the issue was mine 😅 I had completely forgotten that there was an issue on 0.3.3. sorry about that. 
issue is fixed in 0.3.3b xd.

dswinton, 2025-03-10:
Haha, no worries!  😆 

If it's fixed in the new version that's awesome! 👍🏻 
I just wanted to raise here so that other people don't run into the same issue - glad it's resolved, thank you! 🙏🏻 

Regarding these two I've reversed polarity on...

I tried powering them up using the surface mount pads (with the correct polarity) but no lights come on on the nice nano.  

It seems strange to me, because I can measure the battery voltage on the NiceNano here...

[image]

But it doesn't power up.

If I power the nano via USB it boots into the Meshtastic firmware.  So the Nano itself is OK.

I don't really understand the rest of the power delivery to the Nano... but I guess one of the voltages it needs on another pin must not be getting to it?

I'll try to take close up photos when I get home tonight 👍🏻

valzzu, 2025-03-10:
> Haha, no worries! 😆
>
> If it's fixed in the new version that's awesome! 👍🏻 I just wanted to raise here so that other people don't run into the same issue - glad it's resolved, thank you! 🙏🏻
>
> Regarding [...]

one person mentioned that they had issues with the promicro after the same thing that happened to u but after swapping a new one on the board everything worked fine.

dswinton, 2025-03-12:
Heya @valzzu !

Sorry for the slow update here... but when you said...
> after swapping a new one on the board everything worked fine.
...I started focussing on the ProMicro.

It was the firmware flash!!  I re-flashed the Meshtastic firmware and... they're both working!
I have no idea _why_ flashing the firmware worked, but it did.

I guess something about reverse polarity into the ProMicro somehow causes it to lose/corrupt the flash.  It had no LED's, which made me think the board was "dead", but I found when I connected USB, the board did show up as a serial device, and sending the DFU command over serial made the storage device show up on the computer.

So - @valzzu - a huge credit to your PCB design!  The reverse polarity protection worked perfectly.  It definitely got very hot, but I guess that's just part of how it works.  As far as I can tell, it saved all the components.. the ProMicro and the LoRa radio both seem to be working fine!

So, to anyone reading this in the future with a similar issue...  The reverse polarity protection on the board works great - but if you have the same ProMicro as me, it may have lost/corrupted the firmware in a reverse polarity scenario.  Just plug it in and reflash it and your board may just work again! 😄 

Thanks again, @valzzu - keep being awesome! 🥳

dswinton, 2025-03-12:
One more point of clarification on my last post for anyone else with the same issue as me -- the full process of reflashing is needed, not just the Meshtastic firmware but also the bootloader!

So step 1 - reflash the bootloader...
https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/update-nrf52-bootloader/

Step 2 - reflash the Meshtastic firwmare...
https://flasher.meshtastic.org/

Good luck, and I hope this works for you too!

valzzu, 2025-03-12:
Thx :)
Ill be making more stuff in the future so keep an eye on it xd

Ill close this issue now :)