valzzu/meshtastic-pcbs #11: V0.4 incorrect part number for C26,C27?

wbrucem, 2025-03-11:
Trying to order a batch of V0.4 boards from PCBWAY, ran into a couple of issues in the BOM... Part numbers for C26/C27 appear to for larger size (603 vs 402). Also a bit of confusion about BLE antenna part number. See their comments in attached spreadsheet.

[T-3P2W958218A-10sets-BOM_1W-meshtastic-node_0.4_no-connectors(2025-03-11).xls](https://github.com/user-attachments/files/19180782/T-3P2W958218A-10sets-BOM_1W-meshtastic-node_0.4_no-connectors.2025-03-11.xls)

valzzu, 2025-03-11:
Are u aware that v0.4 might not work at all? 
Are u ready to spend $400 on pcbs that might not work at all?

i've fixed C26 and C27

antenna is 2450AT42E010BE

valzzu, 2025-03-12:
Well do report if it works 😅

wbrucem, 2025-03-12:
Sorry to keep bothering you... C7,C12,C13,C14 are listed as 0402 footprint but that part number maps to a 3216 footprint.

PCBWAY is quoting at least a month lead time (ouch!), due to parts availability. I'm now trying JLC and they spotted this issue. I will definitely keep you informed, assuming I can get the boards built.

valzzu, 2025-03-12:
God dang it, ill check that tomorrow when i get on a pc and fix the issue.
Tbh one reason i haven't yet ordered the boards was the money i would have to spend 😅
Plus currently serving in the military so ye.

valzzu, 2025-03-13:
C7,C12,C13,C14 has no part number listed so idk where its getting that footprint.

[image]

wbrucem, 2025-03-13:
Ah, OK, they just did not make a good selection. I found a couple of others with similar issues for which I was able to pick alternative components.

One final issue, the orientation for U12 is incorrect. Pick and Place currently says "180" rotation, I believe it should be "270". I have already corrected my copy and submitted the order. I'll let you know if anything further is discovered.

Thank you, thank you, thank you!

valzzu, 2025-03-15:
Btw @wbrucem if u are in meshtastics discord i have 2 channels for WashTastic.

Thread where most of the talking happens 😅
https://discord.com/channels/867578229534359593/1315666061416464465

WashTastic projects channel.
https://discord.com/channels/867578229534359593/1346425369246302258